bitaxeorg/bitaxeGamma #2: Wrong display connector

marplukiwi, 2024-08-27:
The display Connector NPTC041KFXC-RC in the BOM File is wron

marplukiwi, 2024-08-27:
[image]

yellowpi, 2024-09-19:
And what is the truth? Did you find out?

yellowpi, 2024-09-19:
> <img alt="Bildschirmfoto 2024-08-27 um 22 12 03" width="295" src="https://private-user-images.githubusercontent.com/151972113/361959241-8443efd0-8d94-4e84-9612-c01000d671b0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY3MjcxOTcsIm5iZiI6MTcyNjcyNjg5NywicGF0aCI6Ii8xNTE5NzIxMTMvMzYxOTU5MjQxLTg0NDNlZmQwLThkOTQtNGU4NC05NjEyLWMwMTAwMGQ2NzFiMC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTE5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkxOVQwNjIxMzdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wMmNjY2Y1NjA1MTU1N2JiMWQ1NDU1NWQ5NGI5ODRhYTg5ZWZhZTRjYTdiZDU2NDc3YjYwMjljMjQ3MDQ3ODFiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.jG4lTepuNS8tF-ERAt03vo_HsVBdXePlcatpmgv4brE">

And what is the truth? Did you find out?

yellowpi, 2024-09-19:
> The display Connector NPTC041KFXC-RC in the BOM File is wron

And what is the truth? Did you find out?

marplukiwi, 2024-09-19:
I soldered the display directly to the PCB without a connector. This is also better because the display only has a small distance to the PCB and cannot bend.

yellowpi, 2024-09-19:
> I soldered the display directly to the PCB without a connector. This is also better because the display only has a small distance to the PCB and cannot bend.

Hello, you can use the screen reinforcement in the link below by printing it from your 3D printer and use it compatible with your screen.
 with love.

link

https://github.com/IamGPIO/Bitaxe201-204-ScreenSaver

skot, 2024-09-19:
Hey, sorry about that. There is no connector needed on the PCB for the display. Get a OLED with male pins and it will friction-fit into the holes on the Bitaxe PCB -- or you can solder them in.

I'll get this fixed on the BOM

yellowpi, 2024-09-19:
> Hey, sorry about that. There is no connector needed on the PCB for the display. Get a OLED with male pins and it will friction-fit into the holes on the Bitaxe PCB -- or you can solder them in.
> 
> I'll get this fixed on the BOM

Mr. skot, is there no need to buy this? Should we buy this component?

skot, 2024-09-19:
J3 part number has been removed from schematic and BOM. There is no need for a part on this footprint.

closed 2024-09-19 by skot