bitaxeorg/bitaxeGamma #4: R2, R3, R4 and R7

Alexgtc, 2024-09-17:
Does anyone know why the list of materials does not show the resistances r2, r3, r4 and r7 and in the gerbers yes? Can the plate be mounted leaving those resistors unmounted?

marplukiwi, 2024-09-19:
According to the circuit diagram, the resistors are not needed
[image]

Alexgtc, 2024-09-19:
Ok, so leaving those resistors without mounting, the gamma bitaxe should work without problems, right?

marplukiwi, 2024-09-19:
If the documentation, i.e. the circuit diagram, is correct then yes. But I cannot confirm this.

Alexgtc, 2024-09-19:
Ok, I'll try to mount it without those resistances to see if it works. Thank you for the help.

yellowpi, 2024-09-19:
@skot dear skot, what do you say about this?

skot, 2024-09-19:
@marplukiwi is correct, R2, R3, R4 and R7 are DNP meaning "Do Not Place".

closed 2024-09-19 by skot

marplukiwi, 2024-09-20:
@skot and R18 DNP or not? R18 is in the BOM but crossed out in the circuit diagram.

skot, 2024-09-20:
The red X in KiCad means DNP