bitaxeorg/bitaxeGamma #9: DFM - Increase pad size of barrel plug and inductor

benjamin-wilson, 2024-10-12:
Increase the pad size of these two elements to provide more solder to the component - the barrel plug can be removed with a strong push of the thumb currently.

4xmajor, 2024-10-15:
I have a few more suggestions/observations regarding these these parts...

1. Due to availability issues, I switched the barrel jack suggested in the schematic (54-00164) out for  PJ-036AH-SMT-TR. 

The pads and contacts are far more substantial than the 54-00164, and it is far more abundantly available (on DigiKey at least)... may be a better choice in general?

2. I also have had to switch out the suggested inductor (SLC1480-301ML) to 104CDMCCDS-R30MC due to availability issues. 

The package size and footprint of the SLC1480-301ML is very awkward. It is basically unique among inductors which satisfy the calculated necessary values for L, Isat and Irms according to the TPS546D24 datasheet (8.2.2.3). (I got: L = 300nH, Irms = 35.12A, Isat(peak) = 40.02A)

The package size of 104CDMCCDS-R30MC seems to be very common for these attributes at 11.5mm x 10mm. This might be a better choice for the schematic as it promotes drop-in replacements and also happens to require larger solder pads for sturdiness.

skot, 2024-10-21:
> I have a few more suggestions/observations regarding these these parts...
> 
> 1. Due to availability issues, I switched the barrel connector suggested in the schematic (54-00164) out for  PJ-036AH-SMT-TR.

funny story; we switched to 54-00164 previously because PJ-036AH-SMT-TR went 100% out of stock everywhere. It's wack-a-mole with these things.

BitMaker-hub, 2024-10-22:
I also think 54-00164 is more standard, we can go furder and find compatible parts to it also. Sure there are more out there

benjamin-wilson, 2024-11-02:
Did this on 602 - sent to fab for verification