bitaxeorg/bitaxeGamma #37: Bitaxe is not detected when connected to a USB-C port

rkuester, 2025-04-28:
A Bitaxe Gamma is not detected when connected to an upstream USB-C port. Presumably, this is due to lack of pull-downs on the Bitaxe's USB-C CC pins. USB Sources detect the attachment of devices through USB-C physical connections by utilizing the CC pins.

I have verified this behavior on both a modern MacBook Air and a Linux server with USB-C ports. Presumably, this would also be an issue with hubs that have USB-C ports facing downstream.

There is no issue when a Bitaxe is connected via a USB-C to USB-A cable, because USB-A ports detect attachment via devices pulling up D+ and or D- (in the Bitaxe's case, internally in the USB controller).

MyOwn2C, 2025-04-28:
Known issue but not fixed. 
Only my oldest Ultra works with USB. 
All my other Supra (2x) and Gamma (3x) none of them worked with USB. 
I just use UART to update if webpage update does not work.

skot, 2025-04-28:
Uuuf, you are totally right. I'm kicking myself for removing these pulldowns a while back. I'll get these put back on in future revs. In the mean time you'll have to use a USB-C to USB-A adapter and cable 😭

I swear I had a USB-C to C cable that magically worked.

skot, 2025-05-12:
- Connect a 5.1 kΩ resistor from CC1 to ground.
- Connect a 5.1 kΩ resistor from CC2 to ground.

skot, 2025-07-03:
I have heard that Bitaxes works on USB-C ports that are not also Thunderbolt ports.

skot, 2025-07-03:
Confirmed adding 5.1k pulldown resistors on CC1 and CC2 on a BitaxeGamma 601 allows it to mount on my MacBook USB-C thunderbolt ports.

skot, 2025-09-10:
Looks like this bug was introduced on 401 hardware. everything prior to that _should_ be unaffected.